RemixVSL/iomemory-vsl #142: OLD original FusionIO card

mxb78, 2025-01-23:
Hello folks, tried to resurrect one of original FusionIO cards back to live. With Fedora41 and kernel 6.12. Compile when fine etc, BUT
modprobe gave a stacktrace(dmesg):

[ 1283.844320] Invalid pltconfig, ensure IPC1 device is enabled in BIOS
[ 1284.010055] iomemory_vsl: loading out-of-tree module taints kernel.
[ 1284.010114] iomemory_vsl: module verification failed: signature and/or required key missing - tainting kernel
[ 1284.024469] ------------[ cut here ]------------
[ 1284.024476] Unpatched return thunk in use. This should not happen!
[ 1284.024483] WARNING: CPU: 1 PID: 28826 at arch/x86/kernel/cpu/bugs.c:3063 __warn_thunk+0x2a/0x40
[ 1284.024499] Modules linked in: iomemory_vsl(OE+) snd_seq_dummy snd_hrtimer snd_seq snd_seq_device snd_timer snd soundcore nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nf_tables qrtr sunrpc binfmt_misc intel_rapl_msr intel_rapl_common intel_pmc_bxt intel_telemetry_pltdrv intel_punit_ipc intel_telemetry_core x86_pkg_temp_thermal intel_powerclamp coretemp ppdev kvm_intel dell_pc platform_profile mei_hdcp ee1004 mei_pxp kvm rfkill dell_wmi dell_smbios sparse_keymap dcdbas wmi_bmof r8169 dell_wmi_descriptor rapl intel_cstate pcspkr mei_me i2c_i801 mei idma64 i2c_smbus realtek parport_pc intel_pmc_core parport intel_vsec pmt_telemetry pmt_class vfat fat loop nfnetlink zram lz4hc_compress lz4_compress xfs i915 mmc_block rpmb_core crct10dif_pclmul i2c_algo_bit crc32_pclmul crc32c_intel drm_buddy polyval_generic sdhci_pci ttm ghash_clmulni_intel
[ 1284.024641]  cqhci sha512_ssse3 sdhci drm_display_helper sha256_ssse3 mmc_core sha1_ssse3 cec ucsi_acpi typec_ucsi video typec wmi pinctrl_geminilake fuse
[ 1284.024668] CPU: 1 UID: 0 PID: 28826 Comm: modprobe Tainted: G           OE      6.12.9-200.fc41.x86_64 #1
[ 1284.024677] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[ 1284.024679] Hardware name: Dell Inc. Wyse 5070 Extended Thin Client/012KND, BIOS 1.1.3 08/19/2018
[ 1284.024683] RIP: 0010:__warn_thunk+0x2a/0x40
[ 1284.024692] Code: 66 0f 1f 00 0f 1f 44 00 00 80 3d d8 bd b8 02 00 74 05 c3 cc cc cc cc 48 c7 c7 20 30 e4 ac c6 05 c3 bd b8 02 01 e8 26 b0 08 00 <0f> 0b c3 cc cc cc cc 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00
[ 1284.024696] RSP: 0018:ffffbb210f3079b8 EFLAGS: 00010282
[ 1284.024701] RAX: 0000000000000000 RBX: ffffffffc1679010 RCX: 0000000000000027
[ 1284.024705] RDX: ffff9bb8f7ca1908 RSI: 0000000000000001 RDI: ffff9bb8f7ca1900
[ 1284.024708] RBP: ffffbb210f307a08 R08: 0000000000000000 R09: 0000000000000000
[ 1284.024711] R10: 746f6e20646c756f R11: 216e657070616820 R12: ffffbb21023107c5
[ 1284.024714] R13: ffffbb210f307a28 R14: 0000562945ba2715 R15: ffffbb210f307ab8
[ 1284.024718] FS:  00007f595590a740(0000) GS:ffff9bb8f7c80000(0000) knlGS:0000000000000000
[ 1284.024722] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1284.024725] CR2: 000056296e774000 CR3: 0000000184624000 CR4: 0000000000352ef0
[ 1284.024730] Call Trace:
[ 1284.024733]  <TASK>
[ 1284.024735]  ? __warn_thunk+0x2a/0x40
[ 1284.024742]  ? __warn.cold+0x93/0xfa
[ 1284.024749]  ? __warn_thunk+0x2a/0x40
[ 1284.024758]  ? report_bug+0xff/0x140
[ 1284.024764]  ? console_unlock+0x9d/0x140
[ 1284.024773]  ? handle_bug+0x58/0x90
[ 1284.024780]  ? exc_invalid_op+0x17/0x70
[ 1284.024785]  ? asm_exc_invalid_op+0x1a/0x20
[ 1284.024792]  ? orc_header+0x33f0/0x33f0 [iomemory_vsl]
[ 1284.024865]  ? __warn_thunk+0x2a/0x40
[ 1284.024871]  ? __warn_thunk+0x2a/0x40
[ 1284.024876]  warn_thunk_thunk+0x1a/0x30
[ 1284.024885]  ? orc_header+0x33f0/0x33f0 [iomemory_vsl]
[ 1284.024954]  init_fio_driver+0x35/0xf0 [iomemory_vsl]
[ 1284.025025]  ? orc_header+0x33f0/0x33f0 [iomemory_vsl]
[ 1284.025093]  ? do_one_initcall+0x58/0x310
[ 1284.025103]  ? do_init_module+0x90/0x260
[ 1284.025111]  ? __do_sys_init_module+0x17a/0x1b0
[ 1284.025121]  ? do_syscall_64+0x82/0x160
[ 1284.025131]  ? __alloc_pages_noprof+0x184/0x330
[ 1284.025141]  ? __mod_memcg_lruvec_state+0xd8/0x1d0
[ 1284.025149]  ? __lruvec_stat_mod_folio+0x83/0xd0
[ 1284.025155]  ? set_ptes.isra.0+0x41/0x90
[ 1284.025162]  ? do_anonymous_page+0xfc/0x8d0
[ 1284.025168]  ? __pte_offset_map+0x1b/0x180
[ 1284.025175]  ? __handle_mm_fault+0xb28/0xf80
[ 1284.025185]  ? __count_memcg_events+0x77/0x130
[ 1284.025192]  ? count_memcg_events.constprop.0+0x1a/0x30
[ 1284.025197]  ? handle_mm_fault+0x21b/0x330
[ 1284.025204]  ? do_user_addr_fault+0x55a/0x7b0
[ 1284.025213]  ? clear_bhb_loop+0x25/0x80
[ 1284.025218]  ? clear_bhb_loop+0x25/0x80
[ 1284.025222]  ? clear_bhb_loop+0x25/0x80
[ 1284.025227]  ? entry_SYSCALL_64_after_hwframe+0x76/0x7e
[ 1284.025235]  </TASK>
[ 1284.025237] ---[ end trace 0000000000000000 ]---
[ 1284.025240] <6>fioinf VSL configuration hash: 8f82ea05bdf1195cb400fb48e4ef09fc49b3c1aa
[ 1284.025306] <6>fioinf
[ 1284.025309] <6>fioinf Copyright (c) 2006-2014 Fusion-io, Inc. (acquired by SanDisk Corp. 2014)
[ 1284.025311] <6>fioinf Copyright (c) 2014-2016 SanDisk Corp. and/or all its affiliates. (acquired by Western Digital Corp. 2016)
[ 1284.025313] <6>fioinf Copyright (c) 2016-2018 Western Digital Technologies, Inc. All rights reserved.
[ 1284.025314] <6>fioinf For Terms and Conditions see the License file included
[ 1284.025316] <6>fioinf with this driver package.
[ 1284.025317] <6>fioinf
[ 1284.025319] <6>fioinf ioDrive driver 6.12.9-415025c-3.2.16.1731         loading...
[ 1284.026454] iodrive 0000:01:00.0: enabling device (0000 -> 0002)
[ 1284.026670] <6>fioinf ioDrive 0000:01:00.0: mapping controller on BAR 5
[ 1284.026819] <6>fioinf ioDrive 0000:01:00.0: MSI enabled
[ 1284.026826] <6>fioinf ioDrive 0000:01:00.0: using MSI interrupts
[ 1284.059658] <6>fioinf ioDrive 0000:01:00.0.0: Starting master controller
[ 1284.992889] <6>fioinf ioDrive 0000:01:00.0.0: Adapter serial number is 52319
[ 1285.927358] <6>fioinf ioDrive 0000:01:00.0.0: Board serial number is 52319
[ 1285.927366] <6>fioinf ioDrive 0000:01:00.0.0: Adapter serial number is 52319
[ 1285.932396] <6>fioinf ioDrive 0000:01:00.0.0: Firmware version 5.0.7 101971 (0x500007 0x18e53)
[ 1285.932403] <6>fioinf ioDrive 0000:01:00.0.0: Platform version 0
[ 1285.932406] <6>fioinf ioDrive 0000:01:00.0.0: Firmware VCS version 1971 [0x800007b3]
[ 1285.932680] <6>fioinf ioDrive 0000:01:00.0.0: Firmware VCS uid 0x8ab107e516ab25e7f95bd9266593c0537468fce3
[ 1285.932686] <3>fioerr ioDrive 0000:01:00.0.0: Firmware version 109322 or greater is required. Current firmware version is 101971.
[ 1285.932690] <3>fioerr ioDrive 0000:01:00.0.0: MINIMAL MODE DRIVER: firmware is incompatible.
[ 1285.936784] <6>fioinf ioDrive 0000:01:00.0: Found device fct0 (Fusion-io ioDrive 320GB 0000:01:00.0) on pipeline 0
[ 1285.937761] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: probed fct0
[ 1285.937793] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Attaching explicitly disabled
[ 1285.937800] <3>fioerr Fusion-io ioDrive 320GB 0000:01:00.0: auto attach failed with error EINVAL: Invalid argument

Would be nice to get this old SLC thingy upp and running as those cards been of a VERY nice quality.
I'm ready for any patch to test as card just sits in lab box.

Br
//mxb

mxb78, 2025-01-23:
I just made a normal 'modprobe <module>' without any extra args

snuf, 2025-01-23:
@mxb78 did you set the kernel options mentioned [here](https://github.com/RemixVSL/iomemory-vsl?tab=readme-ov-file#important-note-for-newer-linux-kernels)

mxb78, 2025-01-23:
not mecking with kernel stuff any more nowadays, can you pls give me CLI with modprobe. or do I really have to mod grub.config and reboot?

snuf, 2025-01-23:
There is no modprobe cli flag, it's a kernel flag

mxb78, 2025-01-23:
ok. I'll try then . thx for quick respond

snuf, 2025-01-23:
@mxb78 the updated FreeBSD driver is [here](https://github.com/RemixVSL/iomemory-vsl-freebsd) :)

mxb78, 2025-01-23:
NO :) I'v been waiting for ZFS on Linux to been in sync with fbsd and stable for some time now. I think linux kernel is better suited to handle ZFS stuff. My ZFS is on linux now.

mxb78, 2025-01-23:
[   14.631368] <6>fioinf ioDrive 0000:01:00.0.0: Board serial number is 52319
[   14.631381] <6>fioinf ioDrive 0000:01:00.0.0: Adapter serial number is 52319
[   14.636381] <6>fioinf ioDrive 0000:01:00.0.0: Firmware version 5.0.7 101971 (0x500007 0x18e53)
[   14.636395] <6>fioinf ioDrive 0000:01:00.0.0: Platform version 0
[   14.636398] <6>fioinf ioDrive 0000:01:00.0.0: Firmware VCS version 1971 [0x800007b3]
[   14.636662] <6>fioinf ioDrive 0000:01:00.0.0: Firmware VCS uid 0x8ab107e516ab25e7f95bd9266593c0537468fce3
[   14.636666] <3>fioerr ioDrive 0000:01:00.0.0: Firmware version 109322 or greater is required. Current firmware version is 101971.
[   14.636670] <3>fioerr ioDrive 0000:01:00.0.0: MINIMAL MODE DRIVER: firmware is incompatible.
[   14.640746] <6>fioinf ioDrive 0000:01:00.0: Found device fct0 (Fusion-io ioDrive 320GB 0000:01:00.0) on pipeline 0
[   14.670346] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: probed fct0
[   14.673564] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Attaching explicitly disabled
[   14.673577] <3>fioerr Fusion-io ioDrive 320GB 0000:01:00.0: auto attach failed with error EINVAL: Invalid argument

mxb78, 2025-01-23:
btw, thx for having this repo. you do a great work to have those nice card keep workig :)

snuf, 2025-01-23:
@mxb78 oh it's your firmware that is far behind. You might want to update it.

mxb78, 2025-01-23:
Q: how :D?

snuf, 2025-01-23:
@mxb78 check [here](https://github.com/RemixVSL/vsl_downloads) and the manual is pinned on Discord if I'm not mistaken. Otherwise you're going to have to find a manual on the firmware update somewhere :)

mxb78, 2025-01-23:
can't we just drop the bar of fw? I might have card of 90G or 60G which might have even older fw

mxb78, 2025-01-23:
So I got .fff file. now README says - discord. whatsup or telegram? or mail me to maxim@unixconn.com (NOT US hosted FYI) :)

snuf, 2025-01-23:
There is an invite link [here](https://github.com/RemixVSL/iomemory-vsl?tab=readme-ov-file#how-to-get-help). Though you just need to follow the manual (either google or get it from the Discord)

mxb78, 2025-01-23:
well, discord was not easy. It was terrible. no cli no other stuff - nothing. On the other hand, I think I could help you. :)
just drop a mail

mxb78, 2025-01-23:
I don't want discord experience any more.

snuf, 2025-01-24:
I know Discord is not optimal. We have a lot of things pinned there that are helpful, and duplicating it to another channel is not optimal either. There's a couple of threads on ServeTheHome about firmware updates that are helpful.

mxb78, 2025-01-27:
Looks like this card is dead after all.
last lines from dmesg:

[28034.959191] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Scanning for old NV data...
[28042.291539] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Average erase count is 0
[28042.291617] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: No old NV data available
[28042.291621] <3>fioerr Fusion-io ioDrive 320GB 0000:01:00.0: No NV data available
[28042.291624] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Init NV data
[28042.291917] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: probed fct0
[28042.294435] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: failed in reading eb headers at EB 3, corrupt eb header (read failed).
[28042.294684] <3>fioerr Fusion-io ioDrive 320GB 0000:01:00.0: Failed to create device: Invalid argument (-22).
[28042.294687] <3>fioerr Fusion-io ioDrive 320GB 0000:01:00.0: auto attach failed with error EINVAL: Invalid argument

I flashed it with the firmware I generated with your shell script.
Guess nothing I can do.

bplein, 2025-01-27:
have you tried `fio-sure-erase` or `fio-sanitize`? I'm not sure if it will work with a card in that state. Those command wipe everything and rewrite the structures. 

[Also, have you followed the instructions and updated your GRUB command line:
](https://github.com/RemixVSL/iomemory-vsl?tab=readme-ov-file#grub)

mxb78, 2025-01-27:
erase give error
root@fedora:~# fio-sure-erase /dev/fct0
WARNING: sanitizing will destroy any existing data on the device!
Do you wish to continue [y/n]? y
Erasing blocks: [                    ] (  0%) \2025 Jan 27 16:47:42 fedora usercopy: Kernel memory overwrite attempt detected to SLUB object 'fusion_user_ll_request' (offset 0, size 3960)!   Erasing blocks: [                    ] (  0%) -

and hangs.

Yes, I'v modified file in /boot/.. for my kernel

here is stacktrace in dmesg efter this usercopy error:

[  346.424158] usercopy: Kernel memory overwrite attempt detected to SLUB object 'fusion_user_ll_request' (offset 0, size 3960)!
[  346.424910] ------------[ cut here ]------------
[  346.424914] kernel BUG at mm/usercopy.c:102!
[  346.425600] Oops: invalid opcode: 0000 [#1] PREEMPT SMP PTI
[  346.426269] CPU: 7 UID: 0 PID: 3466 Comm: fio-sure-erase Tainted: P S      W  OE      6.12.8-100.fc40.x86_64 #1
[  346.426960] Tainted: [P]=PROPRIETARY_MODULE, [S]=CPU_OUT_OF_SPEC, [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[  346.427655] Hardware name: Supermicro Super Server/X10SRL-F, BIOS 3.4 06/09/2021
[  346.428364] RIP: 0010:usercopy_abort+0x78/0x7a
[  346.429085] Code: 51 89 dd 82 eb 0e 48 c7 c2 50 00 e0 82 48 c7 c7 be b4 dc 82 56 48 89 fe 48 c7 c7 e0 91 e7 82 51 48 89 c1 41 52 e8 b8 91 fe ff <0f> 0b 89 c2 be 00 10 00 00 48 c7 c7 60 62 a4 83 e8 03 f3 82 ff e9
[  346.430634] RSP: 0018:ffffb2488fc579d0 EFLAGS: 00010246
[  346.431424] RAX: 0000000000000071 RBX: 0000000000000000 RCX: 0000000000000000
[  346.432215] RDX: 0000000000000000 RSI: ffff9e9bbfba1900 RDI: ffff9e9bbfba1900
[  346.433014] RBP: 0000000000000f78 R08: 0000000000000000 R09: 0000000000000000
[  346.433819] R10: ffff9e7cc164c000 R11: ffffb24880e80020 R12: ffff9e7cccdb0c00
[  346.434633] R13: 0000000000000000 R14: 00007efe6faf9d00 R15: ffff9e7ce76a0000
[  346.435454] FS:  00007efe6fafb6c0(0000) GS:ffff9e9bbfb80000(0000) knlGS:0000000000000000
[  346.436291] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  346.437121] CR2: 00007efe6faf9d00 CR3: 00000001e0622001 CR4: 00000000001726f0
[  346.437961] Call Trace:
[  346.438766]  <TASK>
[  346.439570]  ? __die_body.cold+0x19/0x27
[  346.440383]  ? die+0x2e/0x50
[  346.441149]  ? do_trap+0xca/0x110
[  346.441895]  ? do_error_trap+0x6a/0x90
[  346.442624]  ? usercopy_abort+0x78/0x7a
[  346.443360]  ? exc_invalid_op+0x50/0x70
[  346.444062]  ? usercopy_abort+0x78/0x7a
[  346.444744]  ? asm_exc_invalid_op+0x1a/0x20
[  346.445417]  ? usercopy_abort+0x78/0x7a
[  346.446069]  __check_heap_object+0xd9/0x110
[  346.446711]  __check_object_size+0x213/0x230
[  346.447338]  kfio_copy_from_user+0x23/0x60 [iomemory_vsl]
[  346.448004]  ? ifio_7e3a3.ed40884c3016d13c6a23dedee8f561fbc42.3.2.16.1731+0x172/0xa70 [iomemory_vsl]
[  346.448694]  ? __rmqueue_pcplist+0xbd/0xeb0
[  346.449312]  ? ifio_9d3a1.b97732a7f3ea403681dcacf7ea3a1257d7d.3.2.16.1731+0x57/0x390 [iomemory_vsl]
[  346.450001]  ? fusion_control_ioctl+0x23aa/0x35e0 [iomemory_vsl]
[  346.450720]  ? post_alloc_hook+0xe1/0x130
[  346.451350]  ? get_page_from_freelist+0x3a2/0x1be0
[  346.451970]  ? avc_has_extended_perms+0x251/0x540
[  346.452580]  ? ioctl_has_perm.constprop.0.isra.0+0xcc/0x120
[  346.453175]  ? warn_thunk_thunk+0x1a/0x30
[  346.453762]  ? fusion_control_ioctl_internal+0x22/0x40 [iomemory_vsl]
[  346.454392]  ? __x64_sys_ioctl+0x94/0xd0
[  346.454993]  ? do_syscall_64+0x82/0x160
[  346.455584]  ? __count_memcg_events+0x77/0x130
[  346.456202]  ? count_memcg_events.constprop.0+0x1a/0x30
[  346.456769]  ? handle_mm_fault+0x21b/0x330
[  346.457324]  ? do_user_addr_fault+0x55a/0x7b0
[  346.457866]  ? exc_page_fault+0x7e/0x180
[  346.458388]  ? entry_SYSCALL_64_after_hwframe+0x76/0x7e
[  346.458914]  </TASK>
[  346.459416] Modules linked in: xt_addrtype xt_nat xt_mark xt_comment veth vhost_net vhost vhost_iotlb tap tun overlay xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables bridge stp llc rfkill sunrpc binfmt_misc intel_rapl_msr intel_rapl_common sb_edac x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm iTCO_wdt intel_pmc_bxt iTCO_vendor_support rapl ixgbe intel_cstate intel_uncore cp210x igb mei_me ioatdma mdio ipmi_ssif i2c_i801 iomemory_vsl(OE) dca mei i2c_smbus lpc_ich joydev acpi_power_meter ipmi_si acpi_ipmi ipmi_devintf ipmi_msghandler acpi_pad loop dm_multipath nfnetlink zram lz4hc_compress lz4_compress xfs mpt3sas crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni polyval_generic ghash_clmulni_intel ast sha512_ssse3 sha256_ssse3 raid_class sha1_ssse3 i2c_algo_bit scsi_transport_sas wmi zfs(POE) spl(OE) scsi_dh_rdac scsi_dh_emc scsi_dh_alua ip6_tables ip_tables fuse
[  346.463806] ---[ end trace 0000000000000000 ]---
[  346.538844] RIP: 0010:usercopy_abort+0x78/0x7a
[  346.539454] Code: 51 89 dd 82 eb 0e 48 c7 c2 50 00 e0 82 48 c7 c7 be b4 dc 82 56 48 89 fe 48 c7 c7 e0 91 e7 82 51 48 89 c1 41 52 e8 b8 91 fe ff <0f> 0b 89 c2 be 00 10 00 00 48 c7 c7 60 62 a4 83 e8 03 f3 82 ff e9
[  346.540685] RSP: 0018:ffffb2488fc579d0 EFLAGS: 00010246
[  346.541291] RAX: 0000000000000071 RBX: 0000000000000000 RCX: 0000000000000000
[  346.541889] RDX: 0000000000000000 RSI: ffff9e9bbfba1900 RDI: ffff9e9bbfba1900
[  346.542490] RBP: 0000000000000f78 R08: 0000000000000000 R09: 0000000000000000
[  346.543089] R10: ffff9e7cc164c000 R11: ffffb24880e80020 R12: ffff9e7cccdb0c00
[  346.543701] R13: 0000000000000000 R14: 00007efe6faf9d00 R15: ffff9e7ce76a0000
[  346.544319] FS:  00007efe6fafb6c0(0000) GS:ffff9e9bbfb80000(0000) knlGS:0000000000000000
[  346.544943] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  346.545577] CR2: 00007efe6faf9d00 CR3: 00000001e0622001 CR4: 00000000001726f0

mxb78, 2025-01-27:
root@fedora:~# fio-status -a

Found 1 ioMemory device in this system
Driver version: 3.2.16 build 1731

Adapter: Single Adapter
	Fusion-io ioDrive 320GB, Product Number:FS1-003-321-CS, SN:52319, FIO SN:52319
	Pseudo Low-Profile ioDIMM Adapter, PN:00119200000
	External Power: NOT connected
	PCIe Bus voltage: avg 11.93V min 11.88V max 11.94V
	PCIe Bus current: avg 0.43A max 0.46A
	PCIe Bus power: avg 5.20W max 5.42W
	PCIe Power limit threshold: 24.75W
	PCIe slot available power: 25.00W
	PCIe negotiated link: 4 lanes at 2.5 Gt/sec each, 1000.00 MBytes/sec total
	Connected ioMemory modules:
	  fct0:	Product Number:FS1-003-321-CS, SN:52319

fct0	Detached
	ioDrive 320GB, Product Number:FS1-003-321-CS, SN:52319
	ioDrive 320GB, PN:00214100701, Alt PN:FS1-SS3-321-CS
	Powerloss protection: protected
	PCI:01:00.0, Slot Number:3
	Vendor:1aed, Device:1005, Sub vendor:1aed, Sub device:1010
	Firmware v7.1.17, rev 116786 Public
	Geometry and capacity information not available.
	Format: not low-level formatted
	PCIe slot available power: 25.00W
	PCIe negotiated link: 4 lanes at 2.5 Gt/sec each, 1000.00 MBytes/sec total
	Internal temperature: 51.68 degC, max 51.68 degC
	Internal voltage: avg 1.00V, max 1.00V
	Aux voltage: avg 2.46V, max 2.46V
	Lifetime data volumes:
	   Physical bytes written: 756,224
	   Physical bytes read   : 12,020,736
	RAM usage:
	   Current: 11,720,000 bytes
	   Peak   : 11,720,000 bytes

bplein, 2025-01-27:
oh, that reminds me that some of the utils like fio-sure-erase are becoming brittle with age (using block calls that are no longer supported), and we don't have source for the utils. So the only way to do fio-sure-erase if I recall correctly is to boot from an earlier version of Linux/kernel, install the driver and run the sure erase. 

See https://github.com/RemixVSL/iomemory-vsl/issues/115#issuecomment-1898709752 and you'll see the same references to SLUB. 

Have you tried fio-format on the device?

snuf, 2025-05-03:
@mxb78 late to the party, but don't know if you were on the discord or not... there is a branch named [slub](https://github.com/RemixVSL/iomemory-vsl/tree/slub) that contains a skip for the SLUB issue, and circumvents the command that triggers it. The command is a status poll the tool does, it seems to not affect the functioning of the sure-erase it seems.

snuf, 2025-05-07:
Closing this.

closed 2025-05-07 by snuf