RemixVSL/iomemory-vsl #144: Driver won't load on Ubuntu 24.04

drew-viles, 2025-05-05:
Hi,

Long time user of this repo so thanks for the work you do getting this functioning!

I made the decision to do a reinstall of a server which was previously running ubuntu 22.04 with a kernel of 5.<something> and this all worked perfectly. 

I've done a fresh install of Ubuntu 24.04 server which is running kernel `6.8.0-59-generic` which in theory should be compatible based on the comment in the README about it being tested up to/on `6.11`.

For the sake of testing, I installed Rocky 9 and the DKMS procedure worked flawlessly;  I can make use of the drives.

However on Ubuntu 24.04 with the aformentioned kernel I just cannot get the driver to load.

```
fio-status

Found 4 ioMemory devices in this system with 2 ioDrive Duos
Driver version: Driver not loaded
```

The module is loaded:
```
lsmod | grep iomemory_vsl
iomemory_vsl         1331200  0
```

In dmesg, I see some notable lines, but I'm not 100% on how to approach this to solve it.
```
dmesg | grep -i "iomemory\|fio"

...
[  263.550738] <6>fioinf ioDrive 0000:05:00.0: Found device fct0 (HP 1.28TB MLC PCIe IO Accelerator 0000:05:00.0) on pipeline 0
[  263.550930] <3>fioerr HP 1.28TB MLC PCIe IO Accelerator 0000:05:00.0: failed to map append request
[  263.550931] <3>fioerr HP 1.28TB MLC PCIe IO Accelerator 0000:05:00.0: request page program 000000003591c47d failed -22

.... It then repeats this error for each fctX until ....
[  286.205803] <3>fioerr ioDrive 0000:12:00.0: Driver probe on pipeline 0 failed with error -11: Try again
[  286.205806] <3>fioerr ioDrive 0000:12:00.0: Master pipeline failed
[  286.205858] <3>fioerr ioDrive 0000:12:00.0: Driver probe failed on all pipelines
```

Is there anything you can reommend here?

I get the same output with dkms, source and the dpkg approach.

snuf, 2025-05-06:
@drew-viles thanks for reporting. Unfortunately I'm unable to replicate the issue:
```
agrant@ubuntu2404:~/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16$ sudo dmesg | grep fio
[   48.308892] <6>fioinf VSL configuration hash: 8f82ea05bdf1195cb400fb48e4ef09fc49b3c1aa
[   48.310334] <6>fioinf 
[   48.310757] <6>fioinf Copyright (c) 2006-2014 Fusion-io, Inc. (acquired by SanDisk Corp. 2014)
[   48.312236] <6>fioinf Copyright (c) 2014-2016 SanDisk Corp. and/or all its affiliates. (acquired by Western Digital Corp. 2016)
[   48.314171] <6>fioinf Copyright (c) 2016-2018 Western Digital Technologies, Inc. All rights reserved.
[   48.315722] <6>fioinf For Terms and Conditions see the License file included
[   48.316983] <6>fioinf with this driver package.
[   48.317770] <6>fioinf 
[   48.318213] <6>fioinf ioDrive driver 6.8.0-59-c0f364f-3.2.16.1731       loading...
[   48.379490] <6>fioinf ioDrive 0000:00:04.0: mapping controller on BAR 5
[   48.384101] <6>fioinf ioDrive 0000:00:04.0: MSI enabled
[   48.384579] <6>fioinf ioDrive 0000:00:04.0: using MSI interrupts
[   48.415695] <6>fioinf ioDrive 0000:00:04.0.0: Starting master controller
[   49.477694] <6>fioinf ioDrive 0000:00:04.0.0: Adapter serial number is 440178
[   50.540851] <6>fioinf ioDrive 0000:00:04.0.0: Board serial number is 443248
[   50.541488] <6>fioinf ioDrive 0000:00:04.0.0: Default capacity        320.000 GBytes
[   50.542168] <6>fioinf ioDrive 0000:00:04.0.0: Default sector size     512 bytes
[   50.542959] <6>fioinf ioDrive 0000:00:04.0.0: Rated endurance         5.00 PBytes
[   50.544292] <6>fioinf ioDrive 0000:00:04.0.0: 85C temp range hardware found
[   50.550564] <6>fioinf ioDrive 0000:00:04.0.0: Firmware version 7.1.17 116786 (0x700411 0x1c832)
[   50.551322] <6>fioinf ioDrive 0000:00:04.0.0: Platform version 10 
[   50.551855] <6>fioinf ioDrive 0000:00:04.0.0: Firmware VCS version 116786 [0x1c832]
[   50.552533] <6>fioinf ioDrive 0000:00:04.0.0: Firmware VCS uid 0xaeb15671994a45642f91efbb214fa428e4245f8a
[   50.555794] <6>fioinf ioDrive 0000:00:04.0.0: Powercut flush: Enabled
[   50.662003] <6>fioinf ioDrive 0000:00:04.0.0: PCIe power monitor enabled (master). Limit set to 24.750 watts.
[   50.662892] <6>fioinf ioDrive 0000:00:04.0.0: Thermal monitoring: Enabled
[   50.663486] <6>fioinf ioDrive 0000:00:04.0.0: Hardware temperature alarm set for 85C.
[   50.670272] <6>fioinf ioDrive 0000:00:04.0: Found device fct0 (Fusion-io ioDrive Duo 640GB 0000:00:04.0) on pipeline 0
[   51.344863] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: probed fct0
[   51.372110] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: sector_size=4096
[   51.372779] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: setting channel range data to [2 .. 2047]
[   51.397455] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: Found metadata in EBs 2018-2018, loading...
[   51.419734] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: setting recovered append point 2018+196116480
[   51.427315] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: Creating device of size 320000000000 bytes with 78125000 sectors of 4096 bytes (300398 mapped).
[   51.430154] fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: Creating block device fioa: major: 251 minor: 0 sector size: 4096...
[   51.432204]  fioa: fioa1
[   51.432596] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:04.0: Attach succeeded.
vagrant@ubuntu2404:~/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.2 LTS"
vagrant@ubuntu2404:~/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16$ uname -a
Linux ubuntu2404 6.8.0-59-generic #61-Ubuntu SMP PREEMPT_DYNAMIC Fri Apr 11 23:16:11 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

Please provide a full `fio-status -a`, and a `dmesg | grep fio` after loading the driver.

The firmware not matching with the device is because it's missing from the BOM. Can you provide an `lspci -nnn | grep Mass` so I can see what device it is?

drew-viles, 2025-05-06:
Sure can! and thanks for taking the time on this. I always attempt to solve these things myself before reporting as 9 times out of 10 it's user error :-D But I'm just unable to get it working with ubuntu 24.04 Rocky 9 - Fine, ubuntu 22.04 fine. 

So I actually have two devices - both ones from a company I used to work at that ws getting rid and I thought, hey - why not!

It wouldn't be the end of the world to just replace the, with SSDs but it's a nice to have for now.

I've dropped everything below, sorry for the wall of text!

```
 fio-status -a

Found 4 ioMemory devices in this system with 2 ioDrive Duos
Driver version: Driver not loaded

Adapter: Dual Adapter
        HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers, Product Number:600282-B21, SN:38422
        ioDrive Duo HL, PN:00190000108
        External Power: connected
        PCIe Bus voltage: avg 11.70V min 11.29V max 11.83V
        PCIe Bus current: avg 0.95A max 1.01A
        PCIe Bus power: avg 11.52W max 11.89W
        PCIe Power limit threshold: Disabled
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 2.5 Gt/sec each, 1000.00 MBytes/sec total
        Connected ioMemory modules:
          12:00.0:      Product Number:600282-B21, SN:454069
          11:00.0:      Product Number:600282-B21, SN:454068

12:00.0 ioDIMM 320, SN:454069
        ioDIMM 320, PN:00277100304
        Located in slot 1 Lower of ioDrive Duo HL SN:38422
        PCI:12:00.0
        Vendor:1aed, Device:1005, Sub vendor:103c, Sub device:178e
        Firmware v7.1.15, rev 110356 Public
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 2.5 Gt/sec each, 1000.00 MBytes/sec total
        Internal temperature: 49.71 degC, max 52.66 degC
        Internal voltage: avg 1.01V, max 1.01V
        Aux voltage: avg 2.45V, max 2.45V

11:00.0 ioDIMM 320, SN:454068
        ioDIMM 320, PN:00277100304
        Located in slot 0 Upper of ioDrive Duo HL SN:38422
        PCI:11:00.0
        Vendor:1aed, Device:1005, Sub vendor:103c, Sub device:178e
        Firmware v7.1.15, rev 110356 Public
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 2.5 Gt/sec each, 1000.00 MBytes/sec total
        Internal temperature: 48.23 degC, max 50.20 degC
        Internal voltage: avg 1.03V, max 1.03V
        Aux voltage: avg 2.46V, max 2.46V

Adapter: Dual Adapter
        HP 1.28TB MLC PCIe IO Accelerator, Product Number:641027-B21, SN:94877
        ioDrive Duo HL, PN:00190000108
        External Power: connected
        PCIe Bus voltage: avg 11.71V min 11.29V max 11.85V
        PCIe Bus current: avg 0.94A max 1.00A
        PCIe Bus power: avg 11.38W max 11.77W
        PCIe Power limit threshold: Disabled
        PCIe slot available power: 75.00W
        PCIe negotiated link: 8 lanes at 5.0 Gt/sec each, 4000.00 MBytes/sec total
        Connected ioMemory modules:
          06:00.0:      Product Number:641027-B21, SN:410133
          05:00.0:      Product Number:641027-B21, SN:410137

06:00.0 ioDIMM 640, SN:410133
        ioDIMM 640, PN:00277100602
        Located in slot 1 Lower of ioDrive Duo HL SN:94877
        PCI:06:00.0
[... 340 lines cut ...]
[   32.568139] <6>fioinf ioDrive 0000:12:00.0.0: PCIe power monitor enabled (slave). Limit set to 24.750 watts.
[   32.568144] <6>fioinf ioDrive 0000:12:00.0.0: Thermal monitoring: Enabled
[   32.568146] <6>fioinf ioDrive 0000:12:00.0.0: Hardware temperature alarm set for 85C.
[   32.573874] <6>fioinf ioDrive 0000:12:00.0: Found device fct3 (HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:12:00.0) on pipeline 0
[   32.574061] <3>fioerr HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:12:00.0: failed to map append request
[   32.574062] <3>fioerr HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:12:00.0: request page program 00000000fef89314 failed -22
[   32.577254] <3>fioerr ioDrive 0000:12:00.0: Driver probe on pipeline 0 failed with error -11: Try again
[   32.577256] <3>fioerr ioDrive 0000:12:00.0: Master pipeline failed
[   32.577307] <3>fioerr ioDrive 0000:12:00.0: Driver probe failed on all pipelines
```

```
lspci -nnn | grep Mass

05:00.0 Mass storage controller [0180]: SanDisk ioDimm3 [1aed:1005] (rev 01)
06:00.0 Mass storage controller [0180]: SanDisk ioDimm3 [1aed:1005] (rev 01)
11:00.0 Mass storage controller [0180]: SanDisk ioDimm3 [1aed:1005] (rev 01)
12:00.0 Mass storage controller [0180]: SanDisk ioDimm3 [1aed:1005] (rev 01)
```

Let me know it I can help/provide anything else!

Don't waste too much time though if it's going to be a pain of course. I'm happy to do any testing my side. I'm not married to these so can flash anything that might need flashing/reseat etc. They are just there as ephemeral fast storage and no important data is on them so can be cleaned/killed etc :-D 

Thanks again!

snuf, 2025-05-07:
Have you set the iommu flags for [grub](https://github.com/RemixVSL/iomemory-vsl?tab=readme-ov-file#grub)?

snuf, 2025-05-07:
The re-seat won't be it. The only thing I can think of is the iommu. Especially as Rocky 9 runs 5.8. I assume you're on intel?

drew-viles, 2025-05-07:
Yeah I'm on intel. I did add that to Rocky 9 and ubuntu 22.02 so I did wonder when you mentioned it if I'd missed the update-grub. I'll log in once my daughter is in bed and check it all over.

drew-viles, 2025-05-07:
```
fio-status

Found 4 ioMemory devices in this system with 2 ioDrive Duos
Driver version: 3.2.16 build 1731

fct0    Attached
        ioDIMM 640, SN:410137
        Located in slot 0 Upper of ioDrive Duo HL SN:94877
        Last Power Monitor Incident: 42 sec
        PCI:05:00.0, Slot Number:17
        Firmware v7.1.15, rev 110356 Public
        640.00 GBytes device size
        Internal temperature: 41.83 degC, max 42.33 degC
        Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
        Contained VSUs:
          fioa: ID:0, UUID:6f9fda08-8705-4d61-9ca1-235dda5b3522

fioa    State: Online, Type: block device
        ID:0, UUID:6f9fda08-8705-4d61-9ca1-235dda5b3522
        640.00 GBytes device size

fct1    Attached
        ioDIMM 640, SN:410133
        Located in slot 1 Lower of ioDrive Duo HL SN:94877
        PCI:06:00.0, Slot Number:17
        Firmware v7.1.15, rev 110356 Public
        640.00 GBytes device size
        Internal temperature: 46.76 degC, max 46.76 degC
        Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
        Contained VSUs:
          fiob: ID:0, UUID:c8bc5b71-00d8-49fd-b53a-11e40bcca59e

fiob    State: Online, Type: block device
        ID:0, UUID:c8bc5b71-00d8-49fd-b53a-11e40bcca59e
        640.00 GBytes device size

...

fct2    Attached
        ioDIMM 320, SN:454068
        Located in slot 0 Upper of ioDrive Duo HL SN:38422
        Last Power Monitor Incident: 31 sec
        PCI:11:00.0, Slot Number:10
        Firmware v7.1.15, rev 110356 Public
        320.00 GBytes device size
        Internal temperature: 42.33 degC, max 42.33 degC
        Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
        Contained VSUs:
          fioc: ID:0, UUID:7773ffa9-21e4-406d-baf0-6d096fe4cb65

fioc    State: Online, Type: block device
        ID:0, UUID:7773ffa9-21e4-406d-baf0-6d096fe4cb65
        320.00 GBytes device size

fct3    Attached
        ioDIMM 320, SN:454069
        Located in slot 1 Lower of ioDrive Duo HL SN:38422
        PCI:12:00.0, Slot Number:10
        Firmware v7.1.15, rev 110356 Public
        320.00 GBytes device size
        Internal temperature: 43.80 degC, max 43.80 degC
        Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
        Contained VSUs:
          fiod: ID:0, UUID:fffd68ed-43e2-402b-856f-c2426eb3feef

fiod    State: Online, Type: block device
        ID:0, UUID:fffd68ed-43e2-402b-856f-c2426eb3feef
        320.00 GBytes device size
```

Sorry for using the issues for rubber ducking 🤦 

I could have swore I ran that, butevidentlyy I did not. I had the change in my `/etc/default/grub` but clearly got distracted with something else and forgot to `update-grub`.

Thanks for being patient with me!

closed 2025-05-07 by drew-viles

snuf, 2025-05-07:
No worries! Happy it worked out and I was able to help.